cyborg5/IRLib #9: Panasonic TV remote not recognised

sticilface, 2015-05-15:
Hi,

I've been using the irdecode script to figure out the codes for a variety of remotes.  They all work, except my panasonic TV.  

It is a relatively new 5yrs old Panasonic tx-l42s10b, can you shed some light on this?

Very good lib though, works flawlessly.  I have it working with an ESP8266, and have a universal remote driven by mqtt!

cyborg5, 2015-05-15:
Send me an email at cy_borg5@cyborg5.com with a dump from the dump sketch. I will see if I can figure out what protocol it's using. Then we can write some custom code for it. I'm extremely interested in your ESP8266 code as well. I have an IoT remote build from an Arduino Yun but my plans were to try to get something working with the ESP 8266 next.

sticilface, 2015-05-15:
That is great thank you.  I’ve currently disassembled the reader breadboard, so will get back onto it at the weekend. 

I should have been a bit clearer.  The IR lib is running on a pro mini and I’m using this bridge to link the ESP and arduino.  Works very well i must say.  I just send the CODE,HEX,LENGTH to whatever topic its subscribed to.  Build two of them now, but the living room one doesn’t work because of the TV...

https://github.com/tuanpmt/espduino <https://github.com/tuanpmt/espduino>

To end the MQTT…  having said that someone has your code running on the ESP, came across this today.  So hot of the press..

http://www.esp8266.com/viewtopic.php?f=11&t=3035&p=17293&hilit=ir#p17293 <http://www.esp8266.com/viewtopic.php?f=11&t=3035&p=17293&hilit=ir#p17293>

Thank you for your help

cyborg5, 2015-05-15:
I'm waiting on the ESP8266 to get back in stock from Adafruit to begin experimenting with it myself. I thought about the possibility of porting the library directly to it since it's a pretty powerful system and you can program it directly from the Arduino IDE. However it's getting to be a hassle support all of these new boards coming out. My long-term plan is to create a stripped-down version of the library that just says the raw send and receive using an ATtiny85 based system like an Adafruit Trinket and then communicating with other boards using I2C or SPI.

 

For my own personal use I will probably using Arduino Micro with the ESP like what you're doing.

 

I don't have experience with MQTT but I see that library you linked also has a REST interface which is what I'm used to doing.

 

Anyway send me that dump when you get a chance.

 

https://github.com/tuanpmt/espduino <https://github.com/tuanpmt/espduino>

http://www.esp8266.com/viewtopic.php?f=11 <http://www.esp8266.com/viewtopic.php?f=11&t=3035&p=17293&hilit=ir#p17293> &t=3035&p=17293&hilit=ir#p17293 <http://www.esp8266.com/viewtopic.php?f=11 <http://www.esp8266.com/viewtopic.php?f=11&t=3035&p=17293&hilit=ir#p17293> &t=3035&p=17293&hilit=ir#p17293>

e-dredon, 2015-05-31:
Hello. I would be very interrested into a port of your great lib on the ESP8266. There is a talk about IRlib on the ESP8266 Arduino Git here : https://github.com/esp8266/Arduino/issues/6. I could not find any reference on a any pin attached to a timer on the esp8266 or if this is supported yet by the ESP8266 Arduino or the official/unofficial ESP SDK. Any progress in your porting effort ?

cyborg5, 2015-05-31:
I just recently obtained an ESP8266 breakout board from Adafruit but I've not had a chance to play with it yet. In general it was going to be my policy not to support anything except 8-bit AVR processors. With all of these new 32 bit ARM and embedded Linux boards coming out, I just don't have the time to wade through data sheets and try to figure out the hardware specific portions necessary to support the boards. I also try not to support anything that I can't personally test myself or have someone whom I trust can test it on the actual hardware.

 

That being said I feel that the ESP8266 board is such a significant piece of hardware that I might be persuaded into supporting it directly. But it's going to be a steep learning curve for me as I have very little experience with Internet of things unless the devices I'm using already have a lot of software readily available for them. I've built projects using the Arduino Yun however that just looks like a standard Arduino Leonardo from the perspective of my software.

 

I've heard talk with someone has pointed Ken Shirriff's IRremote library (upon which mine is based) to the ESP8266 however it was using bit-bang to modulate the output signal. Although I experimented with bit-bang, my tests showed that it is only barely accurate enough and may not work in all instances.

 

My plan is to make a stripped-down version of my library that would run on some sort of ATtiny85-based board such as the Adafruit trinket. It would then communicate to these more advanced 32 bit and Linux boards using either SPI or I2C. That would save me from having to support all of these new boards natively.

 

However as I said, I'm open to supporting the ESP8266. If someone in the community was more familiar with the board than I am would like to contact me to help with the coding I would love to hear from them. IRLib basically needs 2 hardware specific features.

 

For receiving signals, we need to call an interrupt routine every 50 µs. Someone who can show me how to do that and we are good to go for receiving. Otherwise we are going to be stuck using the IRrecvLoop receiver which means you can't do anything else while you're waiting for a signal to come in. That severely limits your capability.

 

For transmitting signals, I need to know how to generate a square wave with a 33% on 66% of duty cycle and a frequency ranging from 36 kHz to 58 kHz. If someone could write me some code that will let me select any frequency in that range with that one third duty cycle then we can do all the output we want.

 

I'm also presuming that the core code for the ESP8266 supports the standard functions "delay(milliseconds)" and "delaymicroseconds (microseconds)".

 

If someone can provide me with all of that then it's a piece of cake from there on. Feel free to contact me directly at cy_borg5@cyborg5.com

e-dredon, 2015-06-02:
I've just downloaded the latest unofficial ESP8266 SDK. A driver has been added In the examples folder. It's based the IR-remote lib too. There is a git repository https://github.com/valkuc/esp8266-ir-remote. According to the readme he uses It use FRC1 timer interrupt to create precise PWM signal with specified frequency. You might contact Valduc for more information. There is no receiver/decoding support (yet). You might also find some interests to this beginner tutorial about GPIO/interrupts on the ESP8266. I'm a noob with RTOS (the last time I played with the beta, I bricked my devboard. It took me a day to recover the bootloader), so I won't go any further for now experimenting his library. I'm already quite busy working on my main project but I keep a close eye on yours ;).

danielsmithsd, 2016-01-16:
I added support for my Panasonic TV (2009 Plasma) and submitted pull request #15 

Feel free to merge separately.

sticilface, 2016-03-13:
This is great... could free up a rPi I'm using just for new panasonic protocol.  
Any reason to not merge?

closed 2018-10-11 by cyborg5