cyborg5/IRLib #10: Support for Teensy 3.1

objectiveSee, 2015-06-23:
I'd like to add support for the Teensy 3.1.  There needs to be a change to `IRLibTimer.h`.  I am not 100% sure which timer to be using, but I started the change. I copied the settings from Teensy 2.0 without understanding it.  Any ideas what I should set `IR_SEND_TIMER4_HS` to?

```c
/* Teensy 3.1 */
#elif defined(__MK20DX256__)
	//__MK20DX128__	// teensy 3.0
  #ifdef CORE_TEENSY
  	#define IR_SEND_TIMER4_HS	10
  #else
  	#error Unrecognized Core
  #endif
```

Teensy Datasheet: https://www.pjrc.com/teensy/

PaulStoffregen, 2015-06-23:
Maybe this can help?
https://forum.pjrc.com/threads/28988-Timers-for-Infrared-Library-on-Teensy-3-1?p=76322&viewfull=1#post76322

cyborg5, 2015-06-23:
It's getting nearly impossible to keep up with all the 32-bit platforms coming along. One of the problems is I don't have access to all these platforms or the time to wade through complicated data sheets to try to figure out what needs to be changed to support them. If someone can provide me with a change that they can guarantee me will work on a particular platform then I will consider merging it into the distributed code.

 

I'm looking at the possibility of creating a stripped-down version of the send and receive raw data portions of the library and implementing it on an ATtiny85 based platform such as the Adafruit trinket. Then have the trinket communicate using I2C or SPI or just plain serial data to the 32 bit boards.

 

For the time being I'm thinking of just making it my policy to only support eight bit AVR based platforms.

 

The section of code that you copied only specifies the output pin number. It doesn't do anything to set up the interrupts for input or set up the PWM frequency for output. That stuff is buried deeper in the code.

PaulStoffregen, 2015-06-23:
This is the code from IRremoteInt.h.  It looks like you're using this same abstraction layer, perhaps with the names changed slightly.   (fwiw, I'm the original author of that abstraction layer in Ken's library... contributed years ago, long before Arduino made anything other than '328 boards and Mega).

    // defines for special carrier modulator timer
    #elif defined(IR_USE_TIMER_CMT)
    #define TIMER_RESET ({                                  \
            uint8_t tmp __attribute__((unused)) = CMT_MSC;  \
            CMT_CMD2 = 30;                                  \
    })
    #define TIMER_ENABLE_PWM     CORE_PIN5_CONFIG = PORT_PCR_MUX(2)|PORT_PCR_DSE|PORT_PCR_SRE
    #define TIMER_DISABLE_PWM    CORE_PIN5_CONFIG = PORT_PCR_MUX(1)|PORT_PCR_DSE|PORT_PCR_SRE
    #define TIMER_ENABLE_INTR    NVIC_ENABLE_IRQ(IRQ_CMT)
    #define TIMER_DISABLE_INTR   NVIC_DISABLE_IRQ(IRQ_CMT)
    #define TIMER_INTR_NAME      cmt_isr
    #ifdef ISR
    #undef ISR
    #endif
    #define ISR(f) void f(void)
    #if F_BUS == 48000000
    #define CMT_PPS_VAL 5
    #else
    #define CMT_PPS_VAL 2
    #endif
    #define TIMER_CONFIG_KHZ(val) ({        \
            SIM_SCGC4 |= SIM_SCGC4_CMT;     \
            SIM_SOPT2 |= SIM_SOPT2_PTD7PAD; \
            CMT_PPS = CMT_PPS_VAL;          \
            CMT_CGH1 = 2667 / val;          \
            CMT_CGL1 = 5333 / val;          \
            CMT_CMD1 = 0;                   \
            CMT_CMD2 = 30;                  \
            CMT_CMD3 = 0;                   \
            CMT_CMD4 = 0;                   \
            CMT_OC = 0x60;                  \
            CMT_MSC = 0x01;                 \
    })
    #define TIMER_CONFIG_NORMAL() ({        \
            SIM_SCGC4 |= SIM_SCGC4_CMT;     \
            CMT_PPS = CMT_PPS_VAL;          \
            CMT_CGH1 = 1;                   \
            CMT_CGL1 = 1;                   \
            CMT_CMD1 = 0;                   \
            CMT_CMD2 = 30;                  \
            CMT_CMD3 = 0;                   \
            CMT_CMD4 = 19;                  \
            CMT_OC = 0;                     \
            CMT_MSC = 0x03;                 \
    })
    #define TIMER_PWM_PIN        5

PaulStoffregen, 2015-06-23:
On Teensy 3.1, there are 12 hardware timers.  They have different names, rather than only numbers like AVR's timers.  The twelve timers FTM0, FTM1, FTM2, LPTRM, CMT, PDB, PIT0, PIT1, PIT2, PIT3, RTC, SysTick.

The code above uses the CMT timer (CMT = Carrier Modulator Timer).  It doesn't have a number, because there's only one of that type of timer.

PaulStoffregen, 2015-06-23:
You can find a full copy of the code here:

https://github.com/PaulStoffregen/Arduino-IRremote

Perhaps the one gotcha may be the addition of TIMER_RESET, near the beginning of both interrupts.  It's on lines 284 and 313:

https://github.com/PaulStoffregen/Arduino-IRremote/blob/master/IRremote.cpp#L284
https://github.com/PaulStoffregen/Arduino-IRremote/blob/master/IRremote.cpp#L313

On the AVR chips, TIMER_RESET is an empty macro that compiles to no actual code.  But many other chips tend to have hardware that requires some sort of software acknowledge to clear the interrupt status.  That's what the TIMER_RESET macro is doing, so the interrupt doesn't keep getting serviced in an infinite loop.

TrippyLighting, 2015-12-08:
Here is a link to a thread on the Teensy forum where a user has posted a copy of IRlib 1.5 that includes the necessary updates to IRlib to work with Teensy 3.0 and 3.1. https://forum.pjrc.com/threads/26070-Updated-modified-Version-from-cyborgs-irlib?highlight=IRlib

I've used it successfully on Teensy 3.0 and 3.1 for decoding (not for sending) IR codes sent by the Apple Aluminum IR remote. I am in  process of addign support for Teensy LC.

If Teensy boards are needed for testing, let me know, and I'll arrange for it.

closed 2018-10-11 by cyborg5