cyborg5/IRLib #18: decodeGeneric receives and dumps correct input, fails to decode a value

lhk, 2016-02-15:
I am trying to implement my own protocol by using sendGeneric.
Therefore I created a list of definitions on both sender and receiver side:
`const int numBits = 2;
const unsigned int Head_Mark = 1500;
const unsigned int Head_Space = 800;
const unsigned int Mark_One = 2000;
const unsigned int Mark_Zero = 1300;
const unsigned int Space_One = Head_Space;
const unsigned int Space_Zero = 700; //ignored
const unsigned int kHz = 56;
const bool Use_Stop = true;`

The signal uses varying marks. The invariant Space_One=Head_Space is fulfilled.
It sends only two bits.

This code is used to send two signals. 
The first is the value 2 with the bits 1 and 0. Therefore the expected output should contain a head, then a mark of 2000, then a mark of 1300.

Here is the serial dump:
`Decoded Unknown(0): Value:0 (0 bits)
Raw samples(8): Gap:3968
Head: m1400  s900
0:m1900 s900	1:m1200 s800		 2:m1900
Extent=9000
Mark  min:1200	 max:1900
Space min:800	 max:900`

The next signal is 3 with two 1 bits. The expected output are two marks of 2000.
`
Decoded Unknown(0): Value:0 (0 bits)
Raw samples(8): Gap:42284
Head: m1400  s900
0:m1900 s900	1:m1900 s900		 2:m1900
Extent=9800
Mark  min:1900	 max:1900
Space min:900	 max:900
`

As you can see, both signals are transmitted correctly.
There are small variations in the lengths of marks and spaces, but they seem negligible.

Nevertheless IRLib doesn't decode the value.
In both cases the method decodeGeneric returns 0 and the value of the decoder is not changed.

How can I fix this ?

I am very sure that my setup is working correctly, since I can transmit values with the built-in protocol `Panasonic_Old` which appears to be the only builtin protocol which transmits at 56kHz. That's the frequency needed by my receiver.

cyborg5, 2016-02-15:
When I first rewrote IRLib from IRremote one of the main features was the

lhk, 2016-02-16:
Thank you for the quick answer.
I reworked my marks and spaces, now with different spaces. Didn't work. I turned on the debugging information and read your code again.
Apparently the receivers RawCount is not necessarily the number of bits that was sent.
For the 2 bits in my two test scenarios the decoder registered 8 samples. For 3 bits that became 10. And 4 bits create 12 raw samples.

Seems like a linear dependence. Is that correct ? RawCount = numBits*2 +4 ?

With these changes the receiver can decode the transmission.

cyborg5, 2016-02-16:
Yes you have one entry is the gap, 2 for the header, and one for the final bit means there are four extra. And it takes two counts per bit Mark and a Space. You actually compute bits from the raw count. In decodeGeneric look for bits= Mode however inside that method the count is a variable called "offset".

closed 2018-10-11 by cyborg5