cyborg5/IRLib #20: Support 8MHz Arduinos

flyte, 2016-03-03:
I believe this library only supports 16MHz Arduinos. I make this assumption based on [this line](https://github.com/cyborg5/IRLib/blob/master/IRLib.cpp#L961)

The Arduino Pro Mini 3.3v runs at 8MHz and just returns values of 0 for each button press. Would it be possible to make this configurable?

flyte, 2016-03-03:
I've tried `#define F_CPU 8000000` in my sketch but it doesn't seem to have an effect.

cyborg5, 2016-03-03:
Changing that value in your sketch probably will not trigger recompilation of the library. Try putting the define in the library itself. If this doesn't work, I will get back to you. I do have some 8 MHz boards that I've been wanting to try out. I will figure it out then.

flyte, 2016-03-04:
Thanks for the response @cyborg5. When I change this value in the library, it works on perhaps roughly every 20 key presses. I've increased the `CLKFUDGE` to `10` which makes it work slightly better, but it's certainly not as reliable as on a 16MHz chip.

Any other ideas on what I could change?

z3t0, 2016-03-11:
Won't there need to be a change in the timer / tick calculation when using 8Mhz? (i could be wrong)

closed 2018-10-11 by cyborg5