cyborg5/IRLib #21: Receiving "ghost" signals

lhk, 2016-03-12:
I have a setup where one arduino is both responsible for sending and receiving. It is connected to an IR-LED and an IR-decoder.

I am aware of the restrictions of the software- e.g. it is unable to send/receive at the same time. I re-enable receiving after each transmission. There are multiple of these arduinos, they send to each other.

There is a strange problem.
If I send an IR-signal, the decoder immediately receives a huge amount of "ghost data".
lots of samples with m -50 and s 750
please note the negative sign for the mark.

This seems to be software side. Even if all receivers are completely covered, the signal is still received.

Do you have any idea what could be happening ?

ElectricRCAircraftGuy, 2016-03-13:
What recv object are you using?

bengtmartensson, 2016-03-13:
I suspect a hardware problem, namely that the board fails to keep the supply voltage to the receiver constant while sending, which introduces the spooky signals. Things to try:

* connect an oscilloscope to these pins to check this hypothesis,
* use another receiver, preferably with improved noise rejection, like TSOP ??4??.
* increase the IR-Led resistor, at least as a test (you ARE using a such, aren't you??)
* solder a small (100 nF) capacitator between plus and minus immediately on the receiver,
* put a somewhat larger capacitator (10 microF) between + and GND on the board.

ElectricRCAircraftGuy, 2016-03-13:
If you're using IRrecvPCI it's a known bug in the code. I have a fork which fixes it. Chris, the author of this library, is working on fixing it here. Again, *if* you're using this receive object.

lhk, 2016-03-14:
@bengtmartensson 
Your tip to add a capacitor to the circuit fixed the problem. Thank you very much.

closed 2016-03-14 by lhk