cyborg5/IRLib #22: won't compile on mega

NigelStanley, 2016-04-09:
I've developed a very straightforward sketch from one of the examples that works fine on an UNO.

I am then trying to use it within a complex sketch on a mega that will provide digital control for home audio.

But simply adding the include IRLIB,h statement causes a compile failure.

These are the relevant bits of the error message:

> E:\My Documents\Arduino\libraries\IRLib-master\IRLib.cpp:679:3: warning: this decimal constant is unsigned only in ISO C90 [enabled by default]
   hash = FNV_BASIS_32;
   ^
E:\My Documents\Arduino\libraries\IRLib-master\IRLib.cpp: In function 'unsigned char Pin_from_Intr(unsigned char)':
E:\My Documents\Arduino\libraries\IRLib-master\IRLib.cpp:927:45: warning: '__progmem__' attribute ignored [-Wattributes]
   const unsigned char PROGMEM attach_to_pin[]= {
                                             ^
C:\Users\Nigel\AppData\Local\Temp\build4700127920706838464.tmp/core.a(Tone.cpp.o): In function `__vector_13':
C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Tone.cpp:538: multiple definition of `__vector_13'
IRLib-master\IRLib.cpp.o:E:\My Documents\Arduino\libraries\IRLib-master/IRLib.cpp:1039: first defined here
c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/4.8.1/../../../../avr/bin/ld.exe: Disabling relaxation: it will not work with multiple definitions

ElectricRCAircraftGuy, 2016-04-09:
Looks like you're having a conflict between the tone library and this library. They both use Timer2 by default. Try using [ToneAC](http://playground.arduino.cc/Code/ToneAC) instead, which uses timer1.

NigelStanley, 2016-04-11:
Thanks. A quick comment out of all tone statements gets it working again. I'll now try your suggested changes.

closed 2018-10-11 by cyborg5