cyborg5/IRLib #24: Problems with Adafruit_NeoPixel on Arduino UNO

theextron, 2017-07-26:
Hello,

I am planning to use IRLib with my IR reciever to control my light shows with LED strips using Adafruit_NeoPixel library.

Sadly, after changes were made by Adafruit_NeoPixel, IRLib is only recieving "0" values.
If I add "irrecv.disableIRIn(); irrecv.enableIRIn(); delay(100);" it seems to work perfectly fine, but smaller delays wont work.
To have a fluid LED animation I need a maximum delay of 30 though, but it seems like 30 is not enough for IRLib to recieve my IR signal.

Is there any workaround or fix so that NeoPixel is not interrupting the listening of IRLib?
I tried everything for the last 8 hours or so and I'm getting tired of not getting anything done.
I kind of want to use IR so I dont have to use hardware buttons, but it doesnt seem to work yet...

Any help would be appreciated.
Thank you very much in advance! 👍 

Daniel

ElectricRCAircraftGuy, 2017-07-27:
Hey Daniel, I don't have time to look into this in detail by any means, but here's two more options to try. If one of these works for you please post back and tell us which one works for you in this case. Be sure to try all of the different receiving options with IRLib by the way. See the examples.

 1. https://github.com/ElectricRCAircraftGuy/IRLib
 2. https://github.com/z3t0/Arduino-IRremote

cyborg5, 2017-07-27:
I will look into this. However note that IRLib is no longer officially supported. All support has been shifted to the newer IRLib2. I can't imagine anything that has changed in the NeoPixel library library that would interfere with IRLib. Which receiver class were you using? IRrecv? IRrecvPCI? I know there have been big changes in NeoPixel related to DMA usage on SAMD 21 but that should not affect Arduino Uno.

theextron, 2017-07-27:
Thanks.

I am just using the IRrecev library.
I dont thing I am using IRLib1 or if so, that would not be intended.

I did some research and it sees to have something to do with some kind of timers or interruptions that are beeing turned on and off with the NeoPixel library, and this is corrupting IR libraries.

For instance, opening IRrecev to listen for IR,
setting an LED color and updating the LED strang afterwards and then trying to decode the recieved IR signal, it is always 0.
I'll post a simple example code in few minutes or hours if I am on my computer. I believe you can try the code without actually having a NeoPixel LED strip.

Mit freundlichen Grüßen
Best regards

Daniel

theextron, 2017-07-27:
I uploaded two examples.
One of them is working, one is not. The explanation for this is in the comments of the file.

https://github.com/theextron/IRlib_with_NeoPixel/tree/master

theextron, 2017-07-27:
By the way I forgot to mention:
The only way to make the results of the decoder work again seems to be:

        irrecv.disableIRIn();
        irrecv.enableIRIn();
        delay(1000);
        irrecv.getResults();
        irDecoder.decode();

So you need to reinitialize the IR reciever, wait a few milliseconds and that way anything captured in those 1000ms seems to be decoded correctly. If you skip the reinitialization it seems to be impossible.

To call that reilitialization for now I used that "0" which I'm getting if it's recieving _**anything**_, so I have to press a random button and then the disired button right after another. I could not find a way to make it work with just **one** keypress.

theextron, 2017-07-27:
Sorry to spam you, but I just found out that I opened the issue on the wrong repo.
It is meant to belong to you [IRLib2](https://github.com/cyborg5/IRLib2), that's the library I used.

cyborg5, 2017-07-27:
We can continue the discussion here just to keep everything together. But that's why I asked which library you are using because this is the old repo.

I've looked at your sample code and the problem is that NeoPixels require very specific timing. Because of this the disabled interrupts during the strip.show() method. Because you have everything in a very tight loop it is constantly updating the strip whether you need it or not. Of course I understand this is a pair down example just to illustrate the problem.

Typically the way such application would work is you would only need to update the strip after a keypress. So putting the setPixelColor and show commands inside the if (getResults ()) statement as you did in your "working" example should be sufficient.

However if you are animating the pixels then of course you need to constantly update it. I was able to get your nonworking example to work much better by putting a "delay(100)" after the strip.show() statement.

Note that the setPixelColor method doesn't screw up anything. It's only the strip.show() that disables interrupts.

I tried to see when the disabling of interrupts was added to the NeoPixel library and I think it's been there a long time. There is an ifndef for the NRF52 that is recent in the area of the interrupt disable and reenable but I think that's because you can't disable interrupts on that platform because it would mess up the wireless connections and besides I think that version uses DMA so it doesn't need to take over control of everything to update the pixels.

So try putting a delay after your show command and see if you get better results.

theextron, 2017-07-27:
Okay, thanks.
Sorry again for the misunderstanding.

The problem is that already a delay of 40 is noticeable in my case as I am working on an music visualizer. A delay of 30 works perfectly fine for the animation while infrared still wont work that way unfortunaly.

Do you have any other idea despite getting a second Arduino?

Mit freundlichen Grüßen
Best regards

Daniel

cyborg5, 2017-07-27:
I don't see any workaround. The timing requirements of NeoPixels are pretty specific and if you really need to update them that frequently then there's just not going to be enough CPU cycles to do much of anything else. You could try a SAMD 21 based system such as Arduino Zero or Adafruit Feather M0. Both of which run at 48 MHz and there is a new version of the NeoPixel library that does DMA on those systems so it doesn't tie up your CPU while updating the pixels.

closed 2017-07-27 by cyborg5

theextron, 2017-07-27:
Oh, so what you're assuming is that it's nothing about the IR sensor being too slow, it's the Arduino itself?
Thank you very much for your help and patience anyways. I might rather get a second Arduino Zero to recieve infrared using your library and giving the LED-Arduino analog input according to the buttons pressed.
The only geuss or hope I had left was that there might be a IR library which works without these "interrupts".