cyborg5/IRLib #26: does not compile on Arduino Wavgat Nano

Wavgatuino, 2019-01-06:
### **_Arduino Wavgat Nano Board_**

custom  #Board
code 

`/* Пример программы из IRLib-библиотеки Arduino для инфракрасного кодирования и декодирования
 * Версия 1.3 Январь 2014
 * Авторское право 2014 Крис Янг http://cyborg5.com
 * На основе оригинального примера эскиза для библиотеки IRremote 
 * Версия 0.11 Сентябрь, 2009
 * Copyright 2009 Кен Ширрифф
 * http://www.righto.com/
 */
/*
 * JVC отправляет повторные коды, идентичные обычным кодам JVC
 * однако у них нет заголовка. Поэтому есть дополнительный параметр
 * это говорит вам, следует ли отправлять в виде исходного кода или в виде повтора.
 *
 * Единственным устройством, которое я должен был протестировать этот протокол, был старый видеомагнитофон JVC. Это будет работать, только если по крайней мере
 * 2 кадра передаются, разделенные 45us "пространства". Все JVC одинаковой длины, поэтому мы используем
 * третий параметр в качестве флага, чтобы сказать, следует ли отправлять заголовок.
 * Пошлите раз с третьим параметром " 1 " после этого задержите около 50 микросекунд и пошлите снова
 * с третьим параметром "0".
 */

#include <IRLib.h>

IRsend My_Sender;

void setup()
{
  Serial.begin(9600);
  {pinMode(13,OUTPUT);
  digitalWrite(13,1);
   delay(500); // 0.5-секундная задержка 
  digitalWrite(13,0);
}

// отправлять код при каждом получении символа из последовательного портаvoid loop() {
  for (int e = 0; e < 999999; e+=10) {
  for (int i = 0; i < 3; i++) {
 
    My_Sender.send(JVC,e,1);   delayMicroseconds (50);
    My_Sender.send(JVC,e,0);   delayMicroseconds (50);
    }
   digitalWrite(13,1);
   delay(500); // 0.5-секундная задержка между каждым сигналом
  digitalWrite(13,0);
  }
    
  
}
`

the compiler writes:
> 

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\User\Desktop\Ардуино\скетчи\libraries -fqbn=hardware:WAV8F:lardu_328e -ide-version=10805 -build-path C:\Users\User\AppData\Local\Temp\arduino_build_208872 -warnings=more -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_247977 -prefs=build.warn_data_percentage=75 -verbose C:\Users\User\Desktop\Ардуино\скетчи\IRsendJVC\IRsendJVC.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\User\Desktop\Ардуино\скетчи\libraries -fqbn=hardware:WAV8F:lardu_328e -ide-version=10805 -build-path C:\Users\User\AppData\Local\Temp\arduino_build_208872 -warnings=more -build-cache C:\Users\User\AppData\Local\Temp\arduino_cache_247977 -prefs=build.warn_data_percentage=75 -verbose C:\Users\User\Desktop\Ардуино\скетчи\IRsendJVC\IRsendJVC.ino
Using board 'lardu_328e' from platform in folder: C:\Program Files (x86)\Arduino\hardware\hardware\WAV8F
Using core 'lgt8f' from platform in folder: C:\Program Files (x86)\Arduino\hardware\hardware\WAV8F
Внимание: platform.txt из ядра 'Arduino AVR Boards' содержит устаревшие recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{build.path}/{archive_file}" "{object_file}", автоматически преобразовано в recipe.ar.pattern="{compiler.path}{compiler.ar.cmd}" {compiler.ar.flags} {compiler.ar.extra_flags} "{archive_file_path}" "{object_file}". Ожидайте обновления ядра.
Detecting libraries used...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_WAV8F   "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\cores\lgt8f" "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\variants\lgt8fx8e" "C:\Users\User\AppData\Local\Temp\arduino_build_208872\sketch\IRsendJVC.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_WAV8F   "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\cores\lgt8f" "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\variants\lgt8fx8e" "-IC:\Users\User\Desktop\РђСЂРґСѓРёРЅРѕ\СЃРєРµС‚С‡Рё\libraries\IRLib-master" "C:\Users\User\AppData\Local\Temp\arduino_build_208872\sketch\IRsendJVC.ino.cpp" -o "nul"
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_WAV8F   "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\cores\lgt8f" "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\variants\lgt8fx8e" "-IC:\Users\User\Desktop\РђСЂРґСѓРёРЅРѕ\СЃРєРµС‚С‡Рё\libraries\IRLib-master" "C:\Users\User\Desktop\РђСЂРґСѓРёРЅРѕ\СЃРєРµС‚С‡Рё\libraries\IRLib-master\IRLib.cpp" -o "nul"
Generating function prototypes...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-g++" -c -g -Os -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics  -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10805 -DARDUINO_AVR_LARDU_328E -DARDUINO_ARCH_WAV8F   "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\cores\lgt8f" "-IC:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\variants\lgt8fx8e" "-IC:\Users\User\Desktop\РђСЂРґСѓРёРЅРѕ\СЃРєРµС‚С‡Рё\libraries\IRLib-master" "C:\Users\User\AppData\Local\Temp\arduino_build_208872\sketch\IRsendJVC.ino.cpp" -o "C:\Users\User\AppData\Local\Temp\arduino_build_208872\preproc\ctags_target_for_gcc_minus_e.cpp"
"C:\Program Files (x86)\Arduino\tools-builder\ctags\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\Users\User\AppData\Local\Temp\arduino_build_208872\preproc\ctags_target_for_gcc_minus_e.cpp"
Компиляция скетча...
[... 60 lines cut ...]
       if( (DeltaTime = (EndTime=micros()) - StartTime) > 10000) { //If it's a very long wait

       ^

C:\Users\User\Desktop\Ардуино\скетчи\libraries\IRLib-master\IRLib.cpp:760:49: warning: 'DeltaTime' may be used uninitialized in this function [-Wmaybe-uninitialized]

     irparams.rawbuf[irparams.rawlen++]=DeltaTime;

                                                 ^

Compiling core...
Using precompiled core
Linking everything together...
"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc" -Wall -Os -Wl,--gc-sections -mmcu=atmega328p  -o "C:\Users\User\AppData\Local\Temp\arduino_build_208872/IRsendJVC.ino.elf" "C:\Users\User\AppData\Local\Temp\arduino_build_208872\sketch\IRsendJVC.ino.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_208872\libraries\IRLib-master\IRLib.cpp.o" "C:\Users\User\AppData\Local\Temp\arduino_build_208872/..\arduino_cache_247977\core\core_hardware_WAV8F_lardu_328e_9eef9531f9a1ab4928a19deffb109630.a" "-LC:\Users\User\AppData\Local\Temp\arduino_build_208872" -lm
C:\Users\User\AppData\Local\Temp\arduino_build_208872/..\arduino_cache_247977\core\core_hardware_WAV8F_lardu_328e_9eef9531f9a1ab4928a19deffb109630.a(main.cpp.o): In function `main':

C:\Program Files (x86)\Arduino\hardware\hardware\WAV8F\cores\lgt8f/main.cpp:105: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

Используем библиотеку IRLib-master в папке: C:\Users\User\Desktop\Ардуино\скетчи\libraries\IRLib-master (legacy)
exit status 1
Ошибка компиляции для платы WAVGAT NANO 3.0.

<

cyborg5, 2020-04-19:
That platform is not supported in IRLib. Furthermore this original IRLib is no longer being maintained and is depreciated. You should use IRLib2 which can be found here. However note that it also does not support your platform.
[https://github.com/cyborg5/IRLib2](url)

closed 2020-04-19 by cyborg5

ElectricRCAircraftGuy, 2020-04-19:
URL is broken; here it is again: https://github.com/cyborg5/IRLib2